tuanpmt/espduino #8: Can't get past esp.ready()

chall3ng3r, 2015-03-11:
Hi,

I'm trying to get the example work, but it does not get past `while(!esp.ready());` line. 

MQTT and REST, both examples stop at this line, and on the debug port there's some random gibberish, something like `B..aC^<kpWp~v|` just after `esp.ready()` call. And I also see small blink of blue LED on ESP module.

I even updated the Arduino lib from pull request posted few hours ago on this repo. Still same result. 

Any pointers where I should look for? wiring? baud rate setting?

tuanpmt, 2015-03-12:
try ```delay(1000)```
maybe reset command take a  long time

tuanpmt, 2015-03-12:
Please update new commit, maybe the reason is CH_PD incorrect

jgarbers, 2015-03-12:
I had the same issue -- in my case it was because I had forgotten that I'd burned the "AT" firmware to the ESP8266 board I was using.  Re-flashing with the *espduino* firmware and fixing that CH_PD issue got me past that.  

Other than that, seeing garbage on the debug port is, I think, most likely to be a baud rate mismatch.

chall3ng3r, 2015-03-12:
Thanks guys for your reply. Gonna re-flash and update the espduino lib and try again. 

I will post here my result soon. Thanks again.

tuanpmt, 2015-03-13:
i've changed proto buffer size of arduino from 1024 to 512 bytes, it's ensure everything better

chall3ng3r, 2015-03-14:
I downloaded new update, re-flashed the espduino firmware and updated lib in arduino. 

BTW, I am using Arduino Mega 2560 and have it wired like this:
[image]

Also, updated the sample with correct Serial port representation. Using Serial as debug port, and Serial1 for ESP8266 module.

Now with everything updated, I see it would stop on `esp.ready()` call. Red and blue both LEDs are full on ESP module. There's one thing I noted, on Mega's 3.3v pin there's actually 3.6v. 

Still trying to get it working.

chall3ng3r, 2015-03-14:
Now getting this:

[image]

chall3ng3r, 2015-03-15:
While researching, I found another firmware NodeMCU, which flashed and started up on first try. For now I gonna play with a bit more. Will try espduino again when I get the time. Thanks for your support, and keep up the good work you're doing on this project.

closed 2015-03-15 by chall3ng3r

Namphibian, 2015-09-29:
According to your wiring in the photo you are connecting the ESP directly to the MEGA? Be carefull ESP is 3.V Mega is 5v.