tuanpmt/espduino #10: Mqtt.publish sends trailing 0 characters in data.

igorspar, 2015-03-17:
Example:

mqtt.publish("/room1/motion", "ON");

Another mqtt client will receive this data: "ON\0\0" (with two trailing zero characters).

Bob-Ran, 2015-03-21:
I have the same issue here. It is caused by the while loop at line 157 in espduino.cpp

  while(pad_len % 4 != 0)
    pad_len++;

This loop makes the length of the payload data always a multiple of 4. Hence your payload is: [O,N,null,null] This causes weird behaviour at the receiving side of the mqtt data. 

I tried to run the code without this loop but then I receive the following error:

ARDUINO: fail to setup mqtt

@tuanpmt is there a fix for this?

edwinmat, 2015-03-23:
Same issue here...     Trying to communicate with OpenHAB via a Mosquitto MQTT broker using an ESP8266 connected to an Arduino. All messages are received on the MQTT bus and in OpenHAB, but only messages with 4, 8, 12, ... length are being processed within OpenHAB (they trigger rules, are  being displayed in the UI). All messages with other message lengths are listed in the OpenHAB logs (events.log) but ignored in rules and the UI --> strange behaviour!

closed 2015-03-23 by tuanpmt

tuanpmt, 2015-03-23:
@Bob-Ran because of exchange protocol between MCU and ESP8266, we need data align 4 bytes. I've add length params for data when publish, it will help ESP8266 determine exact length of data to send. Now you can try the fixes

commit 86c26449 (2015-03-23): fixed #9 and resolve #10