tuanpmt/espduino #12: error!!

aryan2013, 2015-04-05:
After Setup, i found this on debug port terminal output:- (Continuously)

Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD: 2, cb: 0, ret: 1, argc: 0
Read CRC: 0690, calculated crc:CMD:

aryan2013, 2015-04-05:
Ok, i followed your reply in https://github.com/tuanpmt/espduino/issues/9 , and followed your instruction,,

i am now successful.

well, few thing not understood, why other subscribe report in debug serial terminal after 3 subscribe and one publish in your demo code. See picture

[image]

tuanpmt, 2015-04-05:
esp_mqtt using a ringbuffer to queue subscribe and publish message, first message is confirm queue to buffer success, after send msg to server, and receive sub ack, you will see subscribe message success

aryan2013, 2015-04-05:
Ok i got it... 

mqtt.subscribe("/topic/0"); //or mqtt.subscribe("topic"); /*with qos = 0*/
  mqtt.subscribe("/topic/1");
  mqtt.subscribe("/topic/2");
  mqtt.publish("/topic/0", "data0");

three subscribe, so they initially queued and then i found three subscribe successful.

Then what about published topic mqtt.publish("/topic/0", "data0");???

also, this topic was subscribed, so it should also returned back to same client.

So, terminal should show one publish confirmation/ack and one receipt of subscribed message. But, there is nothing like that in terminal.

one more query: how to publish and subscribe for QoS2?

aryan2013, 2015-04-06:
tuanpmt, 

Please guide me...

i am really not getting mqtt.publish , response on debug port after soon after mqtt.publish("uptime", "test"); is as below:-

CMD: 7, cb: 0, ret: 0, argc: 5
Arg[0], len: 4:B8-34-FF-3F-

Arg[1], len: 8:75-70-74-69-6D-65-00-00-

Arg[2], len: 4:74-65-73-74-

Arg[3], len: 4:00-00-00-00-

Arg[4], len: 4:00-00-00-00-

Read CRC: 667C, calculated crc: 667C

ibrahimisim, 2015-10-15:
Hello,

When I upload software to esp8266 I get an error that is below. Please help me !!!.

pi@raspberrypi /usr/share/arduino/libraries/espduino $ sudo esp8266/tools/esptool.py -p /dev/ttyUSB0 write_flash 0x00000 esp8266/release/0x00000.bin 0x40000 esp8266/release/0x40000.bin
 Could not find platform dependent libraries 
 Consider setting $PYTHONHOME to [:]
 Traceback (most recent call last):
 File "esp8266/tools/esptool.py", line 22, in 
 import serial
 ImportError: No module named serial

bigpaulie, 2015-10-15:
@ibrahimisim stop posting and just install the pySerial library . Just run sudo pip install pyserial or sudo apt-get install python-serial

ibrahimisim, 2015-10-15:
pi@raspberrypi /usr/share/arduino/libraries/espduino/esp8266/tools $  sudo apt-get install python-serial
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-serial is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 34 not upgraded.

bigpaulie, 2015-10-15:
@ibrahimisim if you type : echo $PYTHONPATH , does it return something ?

ibrahimisim, 2015-10-15:
pi@raspberrypi /usr/share/arduino/libraries/espduino/esp8266/tools $ echo $PYTHONPATH

pi@raspberrypi /usr/share/arduino/libraries/espduino/esp8266/tools $

I saw nothing

ibrahimisim, 2015-10-15:
pi@raspberrypi /usr/share/arduino/libraries/espduino/esp8266/tools $ sudo apt-get install python
Reading package lists... Done
Building dependency tree
Reading state information... Done
python is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 34 not upgraded.
pi@raspberrypi /usr/share/arduino/libraries/espduino/esp8266/tools $

bigpaulie, 2015-10-15:
@ibrahimisim ,OK no need to worry. just type whereis python
it should return something like : 
python: /usr/bin/python /usr/bin/python2.7 /usr/bin/python2.6 /etc/python /etc/python2.7 /etc/python2.6 /usr/lib/python2.7 /usr/lib/python2.6 /usr/bin/X11/python /usr/bin/X11/python2.7 /usr/bin/X11/python2.6 /usr/local/lib/python2.7 /usr/local/lib/python2.6 /usr/include/python2.7 /usr/include/python2.6 /usr/share/python

just export the variables like any other environment variable
nano ~/.bashrc

export PYTHONHOME=/usr/bin/python
export PYTHONPATH=/usr/lib/python2.7

it should work 

you can do this also in /etc/rc.local above the "exit 0" line
if you write it in rc.local it will be available system wide in .bashrc it will only be available for "pi" user

ibrahimisim, 2015-10-15:
pi@raspberrypi /usr/share/arduino/libraries/espduino $ cd  /etc/rc.local
-bash: cd: /etc/rc.local: Not a directory
pi@raspberrypi /usr/share/arduino/libraries/espduino $

ibrahimisim, 2015-10-15:
pi@raspberrypi /usr/share/arduino/libraries/espduino $ echo $PYTHONPATH                           
 /usr/lib/python2.7
pi@raspberrypi /usr/share/arduino/libraries/espduino $

ibrahimisim, 2015-10-15:
pi@raspberrypi /usr/share/arduino/libraries/espduino $ esp8266/tools/esptool.py -p /dev/ttyUSB0 write_flash 0x00000 esp8266/release/0x00000.bin 0x40000 esp8266/release/0x40000.bin
Traceback (most recent call last):
  File "esp8266/tools/esptool.py", line 22, in <module>
    import serial
ImportError: No module named serial
pi@raspberrypi /usr/share/arduino/libraries/espduino $

ibrahimisim, 2015-10-15:
pi@raspberrypi /usr/share/arduino/libraries/espduino $ esp8266/tools/esptool.py -p /dev/ttyUSB0 write_flash 0x00000 esp8266/release/0x00000.bin 0x40000 esp8266/release/0x40000.bin
  File "/usr/lib/python3.2/site.py", line 165
    file = sys.stderr)
         ^
SyntaxError: invalid syntax
pi@raspberrypi /usr/share/arduino/libraries/espduino $

bigpaulie, 2015-10-15:
@ibrahimisim , you are not paying attention .
/etc/rc.local is not a directory to edit it type : sudo nano /etc/rc.local

these settings should work fine on Raspberry Pi (Raspbian) 
export PYTHONHOME=/usr/bin/python
export PYTHONPATH=/usr/lib/python2.7

I've tested these settings and all is well .

ibrahimisim, 2015-10-15:
I did like as you say but I get an error below.

pi@raspberrypi /usr/share/arduino/libraries/espduino $ esp8266/tools/esptool.py                           -p /dev/ttyUSB0 write_flash 0x00000 esp8266/release/0x00000.bin 0x40000 esp8266/                          release/0x40000.bin
Traceback (most recent call last):
  File "esp8266/tools/esptool.py", line 22, in <module>
    import serial
ImportError: No module named serial

pi@raspberrypi /usr/share/arduino/libraries/espduino $ echo $PYTHONPATH                         
 /usr/lib/python2.7
pi@raspberrypi /usr/share/arduino/libraries/espduino $

bigpaulie, 2015-10-15:
You have Python 3.x ... just use 2.7 and it will be fine, now .. it's very late and I will not be able to respond anymore also this is not the right place to do this. uninstall your python 3.x, install python 2.7, reinstall your serial library it should work ... @tuanpmt sorry for the big mess !

ibrahimisim, 2015-10-15:
Thanks a lot for your quick reply. I will try and inform you tomorrow. Good night.

ibrahimisim, 2015-10-15:
I can not remove python.

pi@raspberrypi ~ $ sudo apt-get remove python python2.7 python3 python3.2
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  blt gir1.2-freedesktop gir1.2-gst-plugins-base-0.10 gir1.2-gstreamer-0.10 libatlas3-base
  libavahi-client-dev libavahi-common-dev libblas3 libelfg0 libgfortran3 libglib2.0-bin
  liblapack3 libmad0 libmikmod2 libpcre3-dev libpcrecpp0 libportmidi0 libpulse-mainloop-glib0
  libqt4-dbus libqt4-declarative libqt4-designer libqt4-help libqt4-script libqt4-scripttools
  libqt4-sql libqt4-sql-mysql libqt4-svg libqt4-test libqt4-xmlpatterns libqtwebkit4
  libsdl-image1.2 libsdl-mixer1.2 libsdl-ttf2.0-0 libsdl1.2debian libsmpeg0 libvorbisfile3
  python3-minimal python3.2-minimal qdbus
Use 'apt-get autoremove' to remove them.
The following packages will be REMOVED:
  doxygen-latex gconf2 gdb gksu gstreamer0.10-gconf gstreamer0.10-plugins-good idle
  idle-python2.7 idle-python3.2 idle3 jackd2 libgksu2-0 libglib2.0-dev
  libgstreamer-plugins-base0.10-dev libgstreamer0.10-dev libpulse-dev libpython2.7
  libpython3.2 python python-central python-dbus python-dev python-gi python-numpy
  python-picamera python-pifacecommon python-pifacedigitalio python-pip python-pkg-resources
  python-pygame python-qt4 python-qt4-dev python-rpi.gpio python-serial python-setuptools
  python-sip python-sip-dev python-smbus python-support python-tk python2.7 python2.7-dev
  python3 python3-dev python3-numpy python3-picamera python3-pifacecommon
  python3-pifacedigital-scratch-handler python3-pifacedigitalio python3-pygame
  python3-rpi.gpio python3-serial python3-tk python3.2 python3.2-dev texlive-extra-utils
0 upgraded, 0 newly installed, 56 to remove and 34 not upgraded.
After this operation, 199 MB disk space will be freed.
Do you want to continue [Y/n]? Y
dpkg: warning: files list file for package 'libpcre3-dev:armhf' missing; assuming package has no                       files currently installed
dpkg: warning: files list file for package 'libgcrypt11:armhf' missing; assuming package has no                       files currently installed
dpkg: warning: files list file for package 'libavformat-dev' missing; assuming package has no fi                      les currently installed
dpkg: unrecoverable fatal error, aborting:
 files list file for package 'libavutil-dev' is missing final newline
E: Sub-process /usr/bin/dpkg returned an error code (2)