tuanpmt/espduino #13: how to check if wifi is connected?

juanprahamma, 2015-04-15:
dear tuanpmt

my problem is esp8266 didn't connect again if i turn off then turn on my router, i
1. how to check my esp wifi connection status, i've been try using waitReturn but return value always false?
2. how can i check payload is published or not? 

sorry for my bad english
thank you

tuanpmt, 2015-04-16:
when you subscribe callback ```esp.wifiCb.attach(&wifiCb)``` it will call ```wifiCb``` everytime wifi status changed. You need to check status like this
```
void wifiCb(void* response)
{
  uint32_t status;
  RESPONSE res(response);

  if(res.getArgc() == 1) {
    res.popArgs((uint8_t*)&status, 4);
    if(status == STATION_GOT_IP) {
      debugPort.println("WIFI CONNECTED");
      wifiConnected = true;

    } else {
      wifiConnected = false;
    }

  }
}

```

juanprahamma, 2015-04-19:
i think, i'll set my device to subscribe, just like published status, but i still thinking what is better way for this
anyway thank so much 
mr tuanpmt

closed 2015-04-19 by juanprahamma