tuanpmt/espduino #14: Document firmware used to build bridge?

abl, 2015-04-15:
Hi!

I was mystified - for awhile, your firmware in this repository was the only thing that I could connect to https://api.parse.com (and others) with.

Eventually (using bsdiff and friends) I figured out that it looks like you built using 0.9.3-patch1 and the fixed libssl.a from http://bbs.espressif.com/viewtopic.php?f=7&t=365

If this is true, it'd be nice to have it in the documents. :)

Thanks!

tuanpmt, 2015-04-16:
Yes, it is true, You can build with sdk 0.9.3 with libssl.a patched or sdk 1.0.1 in your link
I think next release SDK included the fixed, so don't need to describe here, only need recommend to build with lasted sdk

abl, 2015-04-17:
Gotcha - I haven't tried 1.0.1 with the new patch yet.

Since new releases of the SDK seem to introduce new bugs, it might be nice to have the SDK used associated with the binaries, but hopefully it won't be interesting in the future.

closed 2015-04-17 by abl