tuanpmt/espduino #15: Cannot change user agent

abl, 2015-04-15:
Setting headers works well, but changing the user agent seems to have no effect.

I'm not sure if this is a bug with esp_bridge or just the particular version compiled and included here.

tuanpmt, 2015-04-16:
i don't have fully tested for this, but with command ```setUserAgent("user agent")```, i've seen header modified in debug code

abl, 2015-04-17:
Hmm. I'll test again with debug - I was using the release binary from this repo. (The debug binary doesn't work for me, but I'm using SoftwareSerial, so I'm really rate limited and lots of other things glitch out.)

n8henrie, 2015-08-27:
I'm using debug code and also cannot set user agent; `rest.setUserAgent("my user agent")` right after setting headers in my code (which work), and I still get `User-Agent: ESPDRUINO@tuanpm`.

n8henrie, 2015-08-27:
Think I got it: https://github.com/tuanpmt/espduino/pull/37

closed 2015-09-01 by tuanpmt