tuanpmt/espduino #16: mqtt issues/debug

prussiap, 2015-04-25:
Hi,
I tried to post in the forums to no avail, also I'm not entirely sure something is broken but here goes. I apologize if this is in the wrong place.

I have espduino flashed on esp01, and arduino nano rev3. I've gotten that combination to work perfectly with restful action, and thingspeak updating. I have also managed to get mqtt to work on the esp01 without an arduino and i've gotten mqtt to work with local clients using cloudmqtt broker.

The problem is i have not managed to get mqtt working with espduino. The debug pins show I connected fine, I use the same script on the readme but when I subscribe I don't see any updated topics. 

To confirm the only changes i made are:

``` !mqtt.begin("DVES_duino", "admin", "Isb_C4OGD4c3", 12 ```

client: I think DVES_duino is client
username: "admin"
password/key (not md5'd?): Isb_C4OGD4c3"  

And I changed this:
```mqtt.connect("yourserver.com", 1883, false); ```
Server, port, 
false stands for ssl or not ? not sure for that one bug guessed. 

Anyway I could have gotten things wrong for sure so wanted to double check and get feedback. Also how can i trigger callbacks or actions on subscribed topic on the arduino from mqtt?

I would really like a way to trigger an action based on something from mqtt parsed. So say a door opened at the entrance triggers a light to turn on an espduino in another room ?

Could somebody elucidate that possibility and thoughts on making that more useable :)..

Any help is much appreciated and thank you for the fantastic esp/duino library. Definitely makes a lot of things possible.

bclouser, 2015-06-03:
I too am having some initial trouble getting this to work, and would appreciate a reply to the above post. I am initially trying to use the public "test.mosquitto.org" to confirm that I can send and receive messages, but so far no dice. Any support, or more example code would be appreciated. And indeed, thank you for this library.

eschlenz, 2015-06-03:
Guys,

I have some sample code that has worked for me. My project is pretty specific in that it monitors topics that some of my Jenkins build servers post messages to. The pertinent code you need is in there though. And I'm pressed for time at the moment, otherwise I would trim it back to just what you guys would need. I thought I would share what I have now in the meantime in the hopes that it helps.

Don't forget to change the SSID and PASS #defines. And for good measure, the email address in the initESP8266() function.

Hopefully this is enough to get you guys past your road blocks. If not, reply here or directly to me and I will find time to refine the code to just what you guys would need.

FYI, I'm not the author of this library. I've noticed that the author hasn't replied yet to most of the questions. So if I can be of further help. just let me know.

````
#include <SoftwareSerial.h>
#include <espduino.h>
#include <mqtt.h>
#include <elapsedMillis.h>
#include <Adafruit_GFX.h>
#include <Adafruit_NeoMatrix.h>
#include <Adafruit_NeoPixel.h>
#include <avr/power.h>

// SSID of WiFi network
#define SSID ""
// Password of WiFi network
#define PASS ""
// Reset pin
#define ESP_RST 		4

#define MQTT_SERVER 	"test.mosquitto.org"
#define MQTT_PORT		1883

#define SERVER_1_NAME 	  "Test Server 1"
#define SERVER_1_LED_NAME "Server 1"
#define SERVER_1_TOPIC 	  "/SomeTopicRoot/Topic1"

#define SERVER_2_NAME 	  "Test Server 2"
#define SERVER_2_LED_NAME "Server 2"
#define SERVER_2_TOPIC 	  "/SomeTopicRoot/Topic2"

#define JENKINS_SUCCESS "SUCCESS"

#define DEBUG_RX 5
#define DEBUG_TX 6
#define DEBUG_BAUD 19200

#define ESP8266_BAUD 19200

#define WIFI_GIVE_UP 30000 // ms

#define LED_PIN            13
#define SCROLL_SPEED 100 // ms

// Struct to describe a Jenkins instance
typedef struct JENKINS_SERVER
{
	const String name;
	const String ledName;
	const String topic;

};

const JENKINS_SERVER SERVERS[] =
{
	{
		SERVER_1_NAME,
		SERVER_1_LED_NAME,
		SERVER_1_TOPIC
	},
	{
		SERVER_2_NAME,
		SERVER_2_LED_NAME,
[... 279 lines cut ...]
	}

}

void mqttPublished(void* response)
{
	debug.println(F("Published"));
}

void resetESP8266()
{
	debug.println(F("Resetting ESP8266..."));
	debug.flush();

	esp.reset();
	delay(5000);  //ESP8266 takes a while to restart
}

void resetArduino()
{
	debug.println(F("Resetting Arduino..."));
	delay(1000);
	setup();
}
````

bclouser, 2015-06-07:
I got it working. I am not really sure what i did. My best guess is that I increased the serial buffer size from 64 to 256. I tried switching it back to 64 (to prove this to be the culprit) and it still worked  :( ... but it seems to be inconsistent.

I appreciate the code you sent @eschlenz.  An intriguing idea! Now all i want to do now is hook up our jenkin's builds at work to an LED marquee!

ibrahimisim, 2015-10-15:
Hello,

When I upload software to esp8266 I get an error that is below. Please help me !!!.

pi@raspberrypi /usr/share/arduino/libraries/espduino $ sudo esp8266/tools/esptool.py -p /dev/ttyUSB0 write_flash 0x00000 esp8266/release/0x00000.bin 0x40000 esp8266/release/0x40000.bin
 Could not find platform dependent libraries 
 Consider setting $PYTHONHOME to [:]
 Traceback (most recent call last):
 File "esp8266/tools/esptool.py", line 22, in 
 import serial
 ImportError: No module named serial

bclouser, 2015-10-15:
Looks like you dont have the Python serial module installed on your
computer.

If you have pip installed, it should be as easy as "pip install pyserial"

-ben

ibrahimisim, 2015-10-15:
pi@raspberrypi /usr/share/arduino/libraries/espduino/esp8266/tools $  sudo apt-get install python-serial
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-serial is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 34 not upgraded.

bclouser, 2015-11-14:
What are we to do with the reset pin. I would think we should tie it to vcc but that seems to make the chip unstable