tuanpmt/espduino #17: How best to handle keepalive timeout?

brunnels, 2015-04-26:
I'm working on a relay that is controlled by MQTT topic.  I want to add a fallback state for each relay so that if the connection to the broker is lost the relay will revert to the fallback state.  I can do this if the broker actively disconnects via the mqtt.disconnectedCb.  What I'm not seeing is a callback if the TCP connection to the broker is lost or if no response is received from the MQTT keepalive.

I'm testing by disabling the network connection on the broker and watching the espduino debug.  I see:

MQTT: Send keepalive packet to 10.0.0.1:1883!
MQTT: Sending, type: 12, id: 0000
TCP: Reconnect to 10.0.0.1:1883
Free memory
TCP: Connect to ip  10.0.0.1:1883
TCP: Reconnect to: 10.0.0.1:1883
TCP: Reconnect to 10.0.0.1:1883

I never see a call to the mqtt.disconnectedCb callback.  What is the best way to accomplish what I'm trying to do?

brunnels, 2015-04-26:
Looking at the mqtt.c source in esp_bridge looks like it might need a new tcpDisconnectedCb that gets called around line 300 in here
```
		if(client->reconnectTick > MQTT_RECONNECT_TIMEOUT) {
			client->reconnectTick = 0;
			client->connState = TCP_RECONNECT;
			system_os_post(MQTT_TASK_PRIO, 0, (os_param_t)client);
		}
```

brunnels, 2015-04-27:
Okay, I went ahead and added the tcpDisconnectedCb into esp_mqtt, esp_bridge, and espduino.  I can send a pull request if it's something you want to add into the base.

ibrahimisim, 2015-10-15:
Hello,

When I upload software to esp8266 I get an error that is below. Please help me !!!.

pi@raspberrypi /usr/share/arduino/libraries/espduino $ sudo esp8266/tools/esptool.py -p /dev/ttyUSB0 write_flash 0x00000 esp8266/release/0x00000.bin 0x40000 esp8266/release/0x40000.bin
 Could not find platform dependent libraries 
 Consider setting $PYTHONHOME to [:]
 Traceback (most recent call last):
 File "esp8266/tools/esptool.py", line 22, in 
 import serial
 ImportError: No module named serial