tuanpmt/espduino #18: Make it compatible with the ESP8266 Arduino IDE

marvinroger, 2015-04-27:
Hi,

Now that it is possible to get an [Arduino-compatible IDE with ESP8266 support](https://github.com/esp8266/Arduino), I think having an Arduino next to the ESP8266 is becoming less and less a necessity. It would be great if espduino would be compatible with that environment. I don't know how much this is going to be difficult to realize, but this could be a real leap for the ESP8266 as MQTT and REST are pretty much the standards for IoT today.

There are alternatives compatible with the Arduino env like [PubSubClient](https://github.com/knolleary/pubsubclient) for example, but it lacks some support for QoS (only QoS 0 and 1 for sub, and only QoS 0 for pub) and SSL is not available.

Thanks for your reading.

sticilface, 2015-05-04:
I was going to second this!  It would be really great to have a library for the arduino ESP environment that  did the job of your esp-bridge!  Is this possible? 

Why am i asking? 

Setting up the wifi, and MQTT servers and such on the arduino is fine when you have access to the IDE and flashing abilities.  But if you want to deploy something, then this is not ideal.  If you had a reciprocal server lib for the arduino-ESP IDE then you can have instant compatibility.  Have you ESP running what it needs to, serve web pages, have set up etc, and then the Arduino libs, to give the arduino POST and MQTT functions... 

Your libs work very well.  in a few hrs, (once i realised flashing 5V 16Mhz binaries to 3.3 8Mhz, does give a success but then you get strange things like 9600 baud when 19200 is set in software serial. oops.

tuanpmt, 2015-05-04:
i have no reason for create Arduino IDE library. A lot of things done around, NodeMCU, micropython, all things is fast and easy. The company want to build their app base on official SDK will more realiable.

ibrahimisim, 2015-10-15:
Hello,

When I upload software to esp8266 I get an error that is below. Please help me !!!.

pi@raspberrypi /usr/share/arduino/libraries/espduino $ sudo esp8266/tools/esptool.py -p /dev/ttyUSB0 write_flash 0x00000 esp8266/release/0x00000.bin 0x40000 esp8266/release/0x40000.bin
 Could not find platform dependent libraries 
 Consider setting $PYTHONHOME to [:]
 Traceback (most recent call last):
 File "esp8266/tools/esptool.py", line 22, in 
 import serial
 ImportError: No module named serial

closed 2020-01-07 by marvinroger