tuanpmt/espduino #22: Arduino Leonardo compatibility ?

renno-bih, 2015-05-22:
What is required to make it work with Arduino Leonardo (ATMega 32U4) ?

dogrocker, 2015-05-22:
@renno-bih, Did you try to use with Leonardo ?

I have no Leonardo on hand, But I think it should work.

renno-bih, 2015-05-22:
Yes, it works :)
Using ```Serial1``` for ESP8266 and ```Serial``` for debugging. Also, make sure to use following after initialization of serial interface:
```
while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }
```

dogrocker, 2015-05-22:
Great!

ibrahimisim, 2015-10-15:
Hello,

When I upload software to esp8266 I get an error that is below. Please help me !!!.

pi@raspberrypi /usr/share/arduino/libraries/espduino $ sudo esp8266/tools/esptool.py -p /dev/ttyUSB0 write_flash 0x00000 esp8266/release/0x00000.bin 0x40000 esp8266/release/0x40000.bin
 Could not find platform dependent libraries 
 Consider setting $PYTHONHOME to [:]
 Traceback (most recent call last):
 File "esp8266/tools/esptool.py", line 22, in 
 import serial
 ImportError: No module named serial