tuanpmt/espduino #24: initiate reboot / check connection

sticilface, 2015-05-25:
Hi,

I've build two ESP8266 - Arduino pro mini devices.  Both identical, running the same code.  

One runs just fine.  The other runs just fine, but then stops responding after a day or two, until i restart it.  

Is there a way for the arduino to check the connection to the wifi / connection to the ESP... and if it is faulty restart the ESP, then the arudino...  ?

brunnels, 2015-05-27:
You can add something in the loop and if the wifi isn't up or you don't get a topic message in a specific timeframe then reboot.

tuanpmt, 2015-05-28:
yes, you can call `boolean ESP::ready()` for test connection between esp8266 and arduino. if timeout, you can reset by CH pin

sticilface, 2015-06-27:
Thanks I will try that.

ibrahimisim, 2015-10-15:
Hello,

When I upload software to esp8266 I get an error that is below. Please help me !!!.

pi@raspberrypi /usr/share/arduino/libraries/espduino $ sudo esp8266/tools/esptool.py -p /dev/ttyUSB0 write_flash 0x00000 esp8266/release/0x00000.bin 0x40000 esp8266/release/0x40000.bin
 Could not find platform dependent libraries 
 Consider setting $PYTHONHOME to [:]
 Traceback (most recent call last):
 File "esp8266/tools/esptool.py", line 22, in 
 import serial
 ImportError: No module named serial