tuanpmt/espduino #26: How to get MAC address?

renno-bih, 2015-05-28:
As this new firmware doesn't work with AT commands, is there any way to get MAC address of this module so I can use it in sketch?
Also, it will be very useful to use mDNS or other ways to find module in network. What are plans for that?

bigpaulie, 2015-05-28:
How about the IP address also ?

Namphibian, 2015-09-29:
I am currently working on a modification of the firmware which will extend the wifiCb response. The response will now include mac and ip address as this information can be used in topics on MQTT side.

ibrahimisim, 2015-10-15:
Hello,

When I upload software to esp8266 I get an error that is below. Please help me !!!.

pi@raspberrypi /usr/share/arduino/libraries/espduino $ sudo esp8266/tools/esptool.py -p /dev/ttyUSB0 write_flash 0x00000 esp8266/release/0x00000.bin 0x40000 esp8266/release/0x40000.bin
 Could not find platform dependent libraries 
 Consider setting $PYTHONHOME to [:]
 Traceback (most recent call last):
 File "esp8266/tools/esptool.py", line 22, in 
 import serial
 ImportError: No module named serial

noxqs, 2015-10-21:
Namphibian, how is it coming along ? I need that mac address (or unique id) too !

TLMNicolas, 2015-12-01:
I would also love to have these extra info in the callback

dexterlabora, 2016-01-14:
Any love on a way to get the MAC or IP address of the ESP8266 via espduino? Working with Cactus Micro Rev2.   (I assume this is related to this thread, as to not hi-jack it ;)

Namphibian, 2016-03-20:
Hi All.

I have made some modifications to the library which exposes the following:

1. MAC Address is now published back when connected to a WIFI access point.
2. New Function To Get Signal Strength Of Access Point.

I am happy to expand this with further API extensions.  I am going to expand the data sent back on call back to include IP Address, Gateway and all information I can for IP Configuration. Once that is done I will submit a pull and then hopefully Master Tuan is happy with my C-Fu!

End of week should be achievable!

gautenk, 2016-04-16:
Hi,
How is this coming?
MAC adress would be great, IP even better.
I would love RSSI value!

G