tuanpmt/espduino #28: power down ESP

rikki78, 2015-06-04:
Hi,

I'm trying to power down the ESP in this great piece of software. Disabling works well, now I'm wondering which steps to take to get the ESP back connected.
I've tried enabling, resetting and connecting the ESP. Can you describe which steps to take?

thanks for helping!

tuanpmt, 2015-06-05:
you can reinitialize same as setup code like this
```
void setup() {
  Serial.begin(19200);
  debugPort.begin(19200);
  esp.enable();
  delay(500);
  esp.reset();
  delay(500);
  while(!esp.ready());

  debugPort.println("ARDUINO: setup mqtt client");
  if(!mqtt.begin("DVES_duino", "admin", "Isb_C4OGD4c3", 120, 1)) {
    debugPort.println("ARDUINO: fail to setup mqtt");
    while(1);
  }

  debugPort.println("ARDUINO: setup mqtt lwt");
  mqtt.lwt("/lwt", "offline", 0, 0); //or mqtt.lwt("/lwt", "offline");

/*setup mqtt events */
  mqtt.connectedCb.attach(&mqttConnected);
  mqtt.disconnectedCb.attach(&mqttDisconnected);
  mqtt.publishedCb.attach(&mqttPublished);
  mqtt.dataCb.attach(&mqttData);

  /*setup wifi*/
  debugPort.println("ARDUINO: setup wifi");
  esp.wifiCb.attach(&wifiCb);

  esp.wifiConnect("DVES_HOME","wifipassword");

  debugPort.println("ARDUINO: system started");
}
```

rikki78, 2015-06-08:
Thank you for helping, now it is working!

closed 2015-06-08 by rikki78