tuanpmt/espduino #29: ESPDuino Connect/Reconnect

rupin, 2015-06-05:
Hi,
I have been using an arduino + ESP-01 module and wired them as depicted on your github page.

Apparently an arduino sketch has been functional from a couple of days, and when I started it today, I get the following error

TCP: Reconnect to: m11.cloudmqtt.com:17478
DNS: found ip 107.22.157.224
TCP: connecting...
TCP: Reconnect to m11.cloudmqtt.com:17478
Free memory
TCP: Connect to domain m11.cloudmqtt.com:17478
TCP: Reconnect to: m11.cloudmqtt.com:17478
DNS: found ip 107.22.157.224
TCP: connecting...
TCP: Reconnect to m11.cloudmqtt.com:17478
Free memory
TCP: Connect to domain m11.cloudmqtt.com:17478
TCP: Reconnect to: m11.cloudmqtt.com:17478
DNS: found ip 107.22.157.224
TCP: connecting...
TCP: Reconnect to m11.cloudmqtt.com:17478
Free memory
TCP: Connect to domain m11.cloudmqtt.com:17478
TCP: Reconnect to: m11.cloudmqtt.com:17478
DNS: found ip 107.22.157.224
TCP: connecting...
TCP: Reconnect to m11.cloudmqtt.com:17478

I could not figure out why it would do this. Is this because some setting has changed on the broker?

eschlenz, 2015-06-11:
@rupin Can you share the code?

rupin, 2015-06-11:
I am using the debug version of the MQTT bridge. I reuploaded code on the ESP-01 very recently.

My arduino code is the same as the mqtt demo given in this repo.

eschlenz, 2015-06-11:
I assume you tried pointing to another MQTT broker, just to rule out an issue with connecting to the one you are pointed  at (m11.cloudmqtt.com)? Did you try pointing to test.mosquitto.org:1833?

sriramvis, 2016-04-28:
I have been trying to use espduino and borrow off the mqtt example.

I tried making the system persistent by making cleansession = 0 in the call to mqtt.begin

But the broker does not accept the client on reconnecting, and instead shows a new client connection. 

We suspect that the persistent connection does not work. Is there something I am missing?
 
Also, which version of Mqtt does this library use?