tuanpmt/espduino #30: Problem receiving multiple messages quickly

hackscribble, 2015-06-10:
Hi Tuan

I am using your ESP8266 firmware with your mqtt library on Arduino Nano.  Using a modified version of the example program (see below), I can send and receive individual messages OK.

However, when I try to receive five messages that have been sent in quick succession from another client program, only the first two are received.  Then I see the following debug message and the program hangs.  

```
Get another published message
TCP: data received 114 bytes
```
Checking the messages at the same time using MQTT.fx client, I can see that all five messages are received OK.

Any idea what the cause might be?

Thanks

Ray

```C++
#include <myPushSwitch.h>

#define SWITCH_PIN                     8

PushSwitch testSwitch(SWITCH_PIN, INPUT_PULLUP, LOW, false);  // Enable internal pullup; use default debounce time; wait for double press

switchEvent sEvent;

#include<SoftwareSerial.h>
#include <espduino.h>
#include <mqtt.h>

SoftwareSerial debugPort(2, 3); // RX, TX
ESP esp(&Serial, &debugPort, 4);
MQTT mqtt(&esp);

boolean wifiConnected = false;
boolean mqttIsConnected = false;
boolean sentFirstMQTT = false;

#define BUFFER_SIZE 50
char buffer[BUFFER_SIZE];

void wifiCb(void* response)
{
  uint32_t status;
  RESPONSE res(response);

  if (res.getArgc() == 1)
  {
    res.popArgs((uint8_t*)&status, 4);
    if (status == STATION_GOT_IP)
    {
      debugPort.println(F("WIFI CONNECTED"));
      wifiConnected = true;
      mqtt.connect("hackscribble.ddns.net", 1883, false);
    }
    else
    {
      debugPort.println(F("WIFI NOT CONNECTED"));
      wifiConnected = false;
      mqtt.disconnect();
    }
  }
}

void mqttConnected(void* response)
{
  debugPort.println(F("MQTT CONNECTED"));
  mqtt.subscribe("hackscribble/controls/query");
  mqtt.subscribe("hackscribble/controls/led");
  mqtt.subscribe("hackscribble/controls/pwm");
  mqttIsConnected = true;
}

void mqttDisconnected(void* response)
{

}

void mqttData(void* response)
{
  RESPONSE res(response);

  debugPort.print(F("Received: topic="));
  String topic = res.popString();
  debugPort.println(topic);

  debugPort.print(F("data="));
  String data = res.popString();
  debugPort.println(data);

  topic.toCharArray(buffer, BUFFER_SIZE);
  if (strstr(buffer, "controls/query"))
  {
    reportSwitchState();
  }
  if (strstr(buffer, "controls/led"))
  {
    data.toCharArray(buffer, BUFFER_SIZE);
    if (strstr(buffer, "true"))
    {
      digitalWrite(13, HIGH);
    }
    else if (strstr(buffer, "false"))
    {
      digitalWrite(13, LOW);
    }
  }
}

void mqttPublished(void* response)
{

}

void reportSwitchState()
{
  if (testSwitch.read() == switchWasPressed)
  {
    mqtt.publish("hackscribble/controls/switch", "closed");
  }
  else if (testSwitch.read() == switchWasReleased)
  {
    mqtt.publish("hackscribble/controls/switch", "open");
  }
}

void setup()
{
  
  Serial.begin(19200);
  debugPort.begin(19200);
  debugPort.println(F("ARDUINO: running test-mqtt\r\n\n"));

  pinMode(13, OUTPUT);

  testSwitch.begin();
  sEvent = testSwitch.update();

  esp.enable();
  delay(500);
  esp.reset();
  delay(500);
  while (!esp.ready())
  {
    // empty
  }

  debugPort.println(F("ARDUINO: setting up mqtt client"));
  if (!mqtt.begin("ESP8266", "hackscribble", "hackscribble", 120, 1))
  {
    debugPort.println("ARDUINO: fail to setup mqtt");
    while (1);
  }

  debugPort.println(F("ARDUINO: setting up mqtt lwt"));
  mqtt.lwt("/lwt", "offline", 0, 0);

  /*setup mqtt events */
  mqtt.connectedCb.attach(&mqttConnected);
  mqtt.disconnectedCb.attach(&mqttDisconnected);
  mqtt.publishedCb.attach(&mqttPublished);
  mqtt.dataCb.attach(&mqttData);

  /*setup wifi*/
  debugPort.println(F("ARDUINO: setting up wifi"));
  esp.wifiCb.attach(&wifiCb);

  esp.wifiConnect("<snip>", "<snip>");

  debugPort.println(F("ARDUINO: system started"));

}

void loop()
{

  esp.process();

  if (wifiConnected)
  {
    // empty
  }

  if (mqttIsConnected)
  {
    if (!sentFirstMQTT)
    {
      mqtt.publish("hackscribble/controls/status", "ESP8266 is connected");
      sentFirstMQTT = true;
    }
  }

  sEvent = testSwitch.update();
  if ((sEvent != switchNoChange) && mqttIsConnected)
  {
    reportSwitchState();
  }

}
```

eschlenz, 2015-06-10:
I too have issues with messages coming in in rapid succession. In my case, the Arduino completely freezes up. I've narrowed the down to the espduino Library (IOTW: not my code). I'm not entirely sure why it happens, but I suspect it may be related to @hackscribble's issue.

tuanpmt, 2015-06-11:
Did you try to increase receive buffer size more than 256 bytes?

eschlenz, 2015-06-11:
@tuanpmt No, I did not try that, but thank you for your reply as it led me to a solution.

I looked around in the espduino code for the default buffer size. I think I found it, and I believe it was 512 bytes (?). Which should have been sufficient for my needs.

In debugging my sketch, I found that my available memory was less than 400 bytes. I rearranged some object allocations to free up memory, and now hover around 900 bytes free. I would assume that the 512 bytes for the buffer are allocated only once. Freeing up more memory, however, DID seem to solve the problem for me.

I ran some more tests to see what happened when I sent multiple messages in rapid succession. My sketch no longer crashes. The worst result I got was a momentary freezing of the loop(), followed by an invalid CRC error. It then recovered on it's own. The messages I receive are not mission critical. It will be rare for me that many messages come in rapid succession, and if the worst I get is a random invalid CRC, so-be-it. Good enough for my needs.

hackscribble, 2015-06-11:
Thanks @tuanpmt for the suggestions about memory.  I tried those but it did not help.  And reducing the size of my topic titles and data made no difference either.

But that go me thinking that maybe I was seeing a timing problem, not a memory problem.  The symptoms were consistent with the code not being able to keep up with the speed at which new messages arrive.

So I looked at the `mqttData` callback.  It had several calls to SoftwareSerial.  Also, I am running the debug version of the firmware, so that sends its own messages to SoftwareSerial.  I commented out a couple of my debug print statements and the problem has gone away.

I'll experiment with the no debug firmware and putting my print statements back in.  But I'm going to close this issue.

Many thanks.

closed 2015-06-11 by hackscribble

eschlenz, 2015-06-11:
@hackscribble Good call. I have quite a few Serial print statements in the same place, and I too am running the debug version of the firmware.

hukm, 2015-07-15:
@tuanpmt , hello!

I have faced with quite a similar problem. I am sending mqtt messages from Arduino with ESP as a bridge and after successful 2-3 messages in a row connection between device and broker is lost. 

I have increased time delay between messages from 1 second to 20 but same situation appeared. I have tried to clear some more space in Arduino sketch but that didn't help either.

Is there any command to clear buffer on ESP because it seems to me that the problem is on that side? Any suggestions?

Thank you!

Namphibian, 2015-09-06:
@hackscribble @eschlenz  @tuanpmt 

Was running into the same issue using a Nano and debug software. Due to the number of sensors I must support I had the hardest time getting memory consumption low. Eventually I also got the CRC error in the loop and then it would process again. Thus it would lock up intermittently and then process again. I tried the following serial buffer sizes 16,32,64,128,256,512, I also removed a lot of debug statements.

I suspect my problem was exasperated due to the amount of other libraries in my sketch.
I suspect the debug firmware and debug output statements might be causing an issue where the serial buffers are causing the arduino to wait for some end of line/end of packets this buffer keeps on getting filled and when the arduino finally get the missing eol character it does a crc check. This crc check fails but since the message is complete the arduino can then processes as per normal.

When I changed over to the production version of the firmware the problem seems to not occur. I will do some more testing.

This is all just a hunch though.

hukm, 2015-09-06:
@Namphibian 
I have switched to production version of firmware and increased both HardwareSerial and SoftwareSerial up to 512 as well. But the problem still occurs after 20 minutes of sending information each second.
As for me it is almost fine but however I want to see completely running device without any mistakes

Namphibian, 2015-09-10:
I was using the ESP-01 variant for initial testing as it is just easier to connect. I have now moved to the ESP-07 variant and this problem seems to have vanished. Well based on my tests now I cannot see the system freezing when sending/receiving messages in a 1 second interval for 15 minutes there was a slow down and things got queued but nothing like the lock ups I saw before. This evidence is only anecdotal and once I have done some more testing I will confirm this.