tuanpmt/espduino #32: Progressive Download Support?

vicatcu, 2015-06-24:
@tuanpmt first nice work and thank you for library! This is more of a question / feature request than an issue, per se.

I'm trying to use the REST module to download a large file (a few hundred kilobytes) by modifying the rest example to do: `rest.get("/test.hex");` in `loop`. Printed to the debug console I get `ARDUINO: Invalid CRC` each time through `loop`. I'm not sure why that is happening, perhaps the rest module is not intended to be used as an arbitrary HTTP client accessing a path. Using the ESP8266 AT software, it works fine.

That aside, if I do `rest.get("/test.hex");` in `loop` and the length of my response is longer than the buffer passed to `getResponse` I only get the part of the data that fits in my buffer (which is great, because buffer overruns are a curse). However, it would be very nice if there were a way to progressively download the entire contents in a progressive manner. Again using the ESP8266 AT software, it works fine.

The use case would be when I know I can't store the entire contents of a download in RAM, but I can buffer chunks in RAM and write them off to an external memory like an SD card of SPI Flash. 

So what do you think? Can it be implemented easily? Is it already implemented and I just don't know how to exercise the implementation? Thanks again!

tuanpmt, 2015-06-26:
only support with data less than buffer size, because of SLIP command must parse before process

rupin, 2016-06-25:
I am using the ESP-link firmware on my ESP8266, it allows me to do some nifty configuration from the webserver that the firmware exposes, in addition to allow wireless programming of the arduino.

I am trying to do something similar, get a json string from openweathermaps api. The ESP-link firmware currently has a data limit of 100 characters, which prevents the whole json string to be available in my Arduino.

Is the SLIP protocol a standard implementation, so I could use the ESPduino firmware with ESP-link?