tuanpmt/espduino #38: Timeout with local http servers (Python, Ruby, Node)

n8henrie, 2015-09-01:
I've been trying to figure this out for several days with no luck: http://stackoverflow.com/questions/32283135/post-to-flask-from-espduino-timing-out/

Basically, I made a simple local server to test the reliability of the espduino. Basically the espduino simply `POST`s a number to the server, the server logs the number, and then the number gets incremented. It seems to be working correctly -- the number gets posted and logged with every request.

However, every request also times out (response code `0` in the debug log and takes 5 seconds (default timeout) to complete). I don't understand why this is happening.

I can use the exact same code (and just change the host) and correctly post to Pushover.net (200 response code) or to Requestb.in (200 response code) without any timeouts. I can even start up nginx on my local computer and post to the exact same app as above and get a 200 response code without any timeouts.

However, I've tried both `POST` and `GET` with a local Python Flask server, with the built-in Python `http.server`, with Ruby's built-in `httpd`, and with Nodejs `http-server`, and all of these time out and get a `0` response code.

I've tried examining everything I can think of down to tcpdump and netcat to try to see why nginx works but all these others won't. Please let me know if you have any suggestions or any idea what is causing the timeouts with everything except nginx.

n8henrie, 2015-09-11:
I've continued doing everything I can to debug this. I've updated my SO thread (linked above) as I go. I think I may have narrowed it down considerably using mitmproxy and tcpdump to carefully inspect the packets going through (though most of it is way over my head). It seems that any time the HTTP response from my local server is getting split into 2 packets, I'm getting the timeout (and status code 0) on the espduino. Any time it's a single packet, it works fine.

I can use either nginx or mitmproxy with the exact same Flask app (or other local server), and without a single change to the code, it will work great and espduino gets `200` responses when used behind either reverse proxy (both of which appear to be sending the entire response in a single packet), and if I turn off the proxy, the same code appears to work on the server side but the espduino returns `0` and times out. The response packets getting split into 2 segments is the only difference I can tell, so it seems like [esp_bridge](https://github.com/tuanpmt/esp_bridge) might be choking on them. I don't really know how to verify that this is the problem, so I haven't started an issue there yet.

I'm way above my head here -- does anybody know enough about reassembling ICMP / TCP packets to help look through the esp_bridge code and see if there might be an issue there?

n8henrie, 2015-09-14:
I think this may actually be an esp_bridge problem. See my issue there (linked in the reference above).

ibrahimisim, 2015-10-15:
Hello,

When I upload software to esp8266 I get an error that is below. Please help me !!!.

pi@raspberrypi /usr/share/arduino/libraries/espduino $ sudo esp8266/tools/esptool.py -p /dev/ttyUSB0 write_flash 0x00000 esp8266/release/0x00000.bin 0x40000 esp8266/release/0x40000.bin
 Could not find platform dependent libraries 
 Consider setting $PYTHONHOME to [:]
 Traceback (most recent call last):
 File "esp8266/tools/esptool.py", line 22, in 
 import serial
 ImportError: No module named serial