tuanpmt/espduino #39: MQTT Topic depth?

Turael, 2015-09-30:
Hey friends,
i got a problem with my MQTT connection. I sent a message through MQTT.fx (Desktop Client) a message to the topic /network/mode (message: 1). On the Arduino i only get this: 
```
TCP: data received 18 bytes
```
nothing more... my callback method isnt called by espduino. If i try to sent a message to /mode then I get a message and my callback method working normaly...
```
TCP: data received 11 bytes
Callback method
Received: topic=/mode..
data=1...
```
 I think there is a problem with multiple level in topics? 
Does someone has the same problem? 
As mqtt broker I use mosquitto (on a unix system) - perhaps i need to configure some more to enable multiple level in topics?

ymaglaras, 2015-10-04:
I had a similar problem with very long topics (i.e. /home/firstfloor/livingroom/windows/left/down etc.), especially when I had many of them (states/commands/sensors, so on). At first I tried shortening the topics, but I still had problems. Apparently you might have lots of SRAM when compiling, but when your program runs on the Arduino, it consumes more SRAM than that and runs out of memory, causing all sorts of problems.

So I decided to put everything in PROGMEM like so:

const char PROGMEM superLongTopic[] = "/super/long/topic/that/goes/on/and/on/and/on/and/on";
(add a bunch of those)

and then you need some function to bring them back from PROGMEM:

#define MAX_STRING 65 //your longest possible topic. Careful, this is actually consuming SRAM
char stringBuffer[MAX_STRING];
char* getString(const char* str) {
  strcpy_P(stringBuffer, (char*)str);
  return stringBuffer;
}

So, when you need your topic, you call getString(superLongTopic)

ibrahimisim, 2015-10-15:
Hello,

When I upload software to esp8266 I get an error that is below. Please help me !!!.

pi@raspberrypi /usr/share/arduino/libraries/espduino $ sudo esp8266/tools/esptool.py -p /dev/ttyUSB0 write_flash 0x00000 esp8266/release/0x00000.bin 0x40000 esp8266/release/0x40000.bin
 Could not find platform dependent libraries 
 Consider setting $PYTHONHOME to [:]
 Traceback (most recent call last):
 File "esp8266/tools/esptool.py", line 22, in 
 import serial
 ImportError: No module named serial

Turael, 2015-10-25:
Hey  ymaglaras,

thanks for your response and you're right. I tried your snippet and it works :)
I still need to learn alot about the arduino first I think... 

So my problem is not a ESPduino problem but my (arduino) problem... :)