tuanpmt/espduino #40: When I upload software to esp8266  I get an error that is blow. Please help me !!!.

ibrahimisim, 2015-10-12:
Hello,

When I upload software to esp8266  I get an error that is blow. Please help me !!!.

pi@raspberrypi /usr/share/arduino/libraries/espduino $ sudo esp8266/tools/esptool.py -p /dev/ttyUSB0 write_flash 0x00000 esp8266/release/0x00000.bin 0x40000 esp8266/release/0x40000.bin
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Traceback (most recent call last):
  File "esp8266/tools/esptool.py", line 22, in <module>
    import serial
ImportError: No module named serial

Namphibian, 2015-10-15:
This is not an issue for espduino. This is a python and flashing issue. Please try resolve the issue on google first. Please post only one question with the error. When you post on a lot of other questions like you did it is really annoying as it creates a lot of noise and clutters the questions.

I ran into the same problems when I started. Fix one problem at a time and try to resolve them either with the pythons and python tool developers. Tuan did NOT create these tools so thus posting here is not going to be that helpful for you.

tim3in, 2016-09-09:
you need to install **Python Serial Port Extension**

i have same issue 
check your python version 
for python 2.7 go to

**https://pypi.python.org/pypi/pyserial/2.7**

hope this will resolve your issue