tuanpmt/espduino #42: HTTPS Rest API

sergione76, 2015-11-25:
Hi,
I'm trying to use https rest api with different servers: with a lot of server doesn't works infact
rest.getResponse returns 0
For example connecting using https with github.com it doesn't work...some ideas or help?
Thank you
Sergio

Namphibian, 2015-11-25:
Hi Sergio.

There is known limitations in the SSL handling routines.

As far as I know it only works with certificates up to 1024 bits in
strength.

Kind Regards
Neil Franken

sergione76, 2015-11-26:
Hi Neil,
yes but...sometimes it works on servers with certificate that have 2048 key length! e.g https://yourapihere-com-r2pgihowjx7x.runscope.net/
Regards
Sergio

Namphibian, 2015-11-26:
Hi Sergio.

My bad mqtt only supports 1024 bit. Will check on this some more over next
few days.

Kind Regards
Neil Franken

guachitonico, 2016-11-22:
Any news on this? :S

Namphibian, 2016-11-22:
Hi People.

Just a reminder that saying that it does not work does not help nobody.
What errors are you seeing? Did you try and debug it? All of these problems
could be resolved by giving developers at least the error message for
example. In summary if you can't give an error then we can't fix the error.

guachitonico, 2016-11-22:
Well, I fixed it after a lot of trial and error.

Couple of things if anybody else has the same problem:

Use rest.begin("HOST",80, false) instead of rest.begin("host");

When GET'ing something, put the response value to exactly the value you're expecting, or more (up to 500 i tried). If you put less, you'll get an invalid CRC error, as well as if you go over 500.

I finally got my sketch to connect to io.adafruit.com and get my feeds!