tuanpmt/espduino #46: Is there any timeline for the WiFi AP portion?

vidia, 2016-01-26:
If not, can you point me in the right direction so I can find out how to do it myself?

Namphibian, 2016-02-01:
Hi David.

I have managed to add the mac address to the wifi connect callback. Thus at
the moment I can retrieve the mac address after the esp connects.

The files you need to study is wifi.c and cmd.c. I am busy testing this and
will probably do a pull and merge soon.

I was hoping to get the wifi api more or less translated into arduino
function calls but my time is limited due to me moving between states.

Contact me if you want some code and to help out. I could use a extra pair
or eyes.

Kind Regards
Neil Franken

thinkingste, 2016-07-05:
Is there any progress on this issue?