tuanpmt/espduino #48: malloc assert!

abhaysbharadwaj, 2016-02-18:
I have been working with this wonderful library.
in my latest project i am trying to send a "RFID number" using esp8266.
I am getting the error "malloc assert!" in my serial window of arduino.

The sequence of events shown in my serial monitor of arduino are as follows:

"ARDUINO: setup wifi
ARDUINO: system started
scandone
add 0
aid 7
pm open phy2,type:2 0 0
cnip:192.168.10.108,mask:255.255.255.0,gw:192.168.10.1
CONNECTED TO WIFI NETWORK
mqtt Connected Successfully
null
malloc assert!"

for all other programs that i have done, i have not encountered this issue. So, i can say its because of RFID that i might be getting this problem.

My question is, is this the issue with esp8266 or the library? because, i am unable to find "malloc assert!" sentence anywhere in the mqtt library. 

sorry for the long question. Hope anyone can help me out! Thanks in advance..

Familienpapi, 2016-10-07:
I have the same problem when I'm doing something more with the mqtt example code. I'm trying to read sensors via I2C, an IR remote control, a motion sensor and sending the results via mqtt to FHEM and a TFT Touch Display (UTFT).
When I comment out some code fragments then it works so I think, it's a timing problem.
I'm using an Arduino Mega2560 R3 and IDE 1.6.12 portable on Windows x64. 

I hope that helps a little. I'm giving up.

Familienpapi

frugallabs, 2016-12-08:
This issue was with the ESP01 module having 512Kb memory (The blue ones)..
The ESP01 available in the market now have 1Mb memory (brown color modules). I am no longer facing issues with these new modules now! :)