tuanpmt/espduino #51: Unable to communicate

lehighkid, 2016-03-16:
I have followed the installation guide (flashed esp w/ firmware, connected arduino uno and esp as shown, uploaded example MQTT sketch to arduino uno) but I receive garbage in the serial port.  I've tried both the FTDI and ardunio port at several different baud rates.  I've tried both the release and debug firmwares.  I know I can use the esp in stand alone but I would like to add wifi to my ardunio.  I'm not sure what else to try...  Do I need the FTDI controller once it is flashed?  Any suggestions?  Any help direction would be greatly appreciated.

Namphibian, 2016-03-16:
Hi Aaron.

From the top of my head here is some pointers.

Just checking but you are running a logic level converter between Arduino
and ESP? If not you might have fried the ESP by now. All connections on ESP
is 3,3v only. You might read about some people being lucky with 5V between
the two but that advise will bring the blue smoke monster.

If there is a logic level converter between the two see below.

Check the following:
1. Tx pin ESP serial to Arduino RX Serial pin.
2. Rx pin ESP serial to Arduino TX pin.
3. Choose pin on Arduino say pin 4 and connect this to the CH-PD on esp.

Initially I had my Tx to Tx and Rx to Rx. Easy mistake to make when you are
wiring even for an old hand.

The CH-PD connection between ESP and Arduino allows Arduino to reset ESP so
that Arduino can control it after start up. The line esp.reset() in code
pulls signal low then high which resets ESP.

Make sure you dont power ESP from 3.3v output on Arduino as the ESP can use
200mA plus and this can kill Arduino. Best is to use a LM1117 or similar
regulator.

I run my custom boards on a 5v power supply with a regulator for 3.3v
regulation for ESP.

If you are using two power supplies ie. 5v for arduino and 3.3v for Esp and
you are using a logic level converter such as sparkfun converter you have
to connect the ground of the 3.3v and 5v circuits. These logic level
converters are normally based on BS138 transistors. If logic level
conversion is ic based, diode based or resistor divider based then you need
to check their application notes for further info.

Kind Regards
Neil Franken

hjfabius, 2016-03-16:
Ciao, unfortunately I'm getting exactly the same behavior of @lehighkid.

My full circuit run around 3.3V using an arduino pro mini.  

 ESP is silent till  `while(!esp.ready());`
and then send approx 10 random chars..   Obviously the code does not proceed further and stay blocked on the `while` loop.

I tried to also reverse Serial and SoftwareSerial but i've no sucess.

An extra power supplier provide up to 600mA... Any further suggestion?

Ciao
hjfabius

lehighkid, 2016-03-16:
@Namphibian Thank you for the list of suggestions and insights.  I do have a logic level converter but I went back to the Nano, started over and was able to get it working... some what - the serial is still garbled but it does connect to Wifi and publish a message on MQTT.  I really appreciate the help - I will definitely save it off for when I attempt to do it again.

NotSirius, 2016-03-17:
I had similar problems until I added pull-up resistors to both of the gpio pins.

hjfabius, 2016-03-17:
Solved also on my side. During the flashing GPIO0 must be low. I forget to remove the low signal during the execution... so it was always in "flashing mode".

Namphibian, 2016-03-17:
Hi Aaron.

Just to clarify you are using hardware serial for ESP and Arduino coms?

If so that should be dedicated to only comms between Arduino and ESP.
Software serial is a hack at best. For example software serial disables
interupt functions during recieves and send, remember it is simulating a
serial port. This can cause hardware serial and other functions to behave
strangely. So use software serial only while debugging and do it sparingly.
If you need more than one hardware serial use the Mega or ATMEL 1284 mcu,
look at the callinium(not sure about spelling) project.

One last thing I must mention. I needed to implement a timer function on my
project using the ESP. For two weeks I struggled with this. When I included
the timer libraries I had some very strange results. Some libraries worked
intermittently other libraries did not even compile. I switched the Arduino
IDE to verbose output and I realised the IDE used the windows temp
directory to store object files.

Since this directory was monitored by my anti virus the anti-virus kicked
in and silently removed, blocked or otherwise mangled the code. I switched
the anti-virus off and compiled and viola code that would not run or even
compile worked like the examples. The Timer2 lib I could not even get am
example to compile.  After adding arduino ide to safe list I never had this
problem again.

So compile your code with anti-virus switched off and check. I have been
coding for 35 years and never had anti-virus softly screw my code up. So
check it.

Kind Regards
Neil Franken