tuanpmt/espduino #52: Help sending multiple publish (MQTT,Bridge)

tolbier, 2016-04-07:
I am using the tuanpmt code & libraries to run an Arduino multi sensor to send mqtt messages through a ESP8266 (Bridge Mode)

The code below runs ok, but the trouble is that it only publishes the message ONCE....(after a certain amount of time, it debugs a "beacon timeout" and it starts to connect and publish again)

I would like to publish maybe each 10 seconds, and keeping alive the Wifi Connection
I don't know exactly how to call the publish method from the loop?, or to stablish a period of time to run this cycle until the arduino is reseted

Could you help me giving me some clues or help?

thanks in advance from Spain.
`
/**
 * \file
 *       ESP8266 MQTT Bridge example
 * \author
 *       Tuan PM <tuanpm@live.com>
 */
#include <SoftwareSerial.h>
#include <espduino.h>
#include <mqtt.h>

SoftwareSerial debugPort(2, 3); // RX, TX
ESP esp(&Serial, &debugPort, 4);
MQTT mqtt(&esp);
boolean wifiConnected = false;

void wifiCb(void* response)
{
  uint32_t status;
  RESPONSE res(response);

  if(res.getArgc() == 1) {
    res.popArgs((uint8_t*)&status, 4);
    if(status == STATION_GOT_IP) {
      debugPort.println("WIFI CONNECTED");
      mqtt.connect("1.0.1.1", 1883);
      //mqtt.connect("1.0.1.1", 1883, false);
      wifiConnected = true;
      //or mqtt.connect("host", 1883); /*without security ssl*/
    } else {
      wifiConnected = false;
      mqtt.disconnect();
    }
  
  }
}

void mqttConnected(void* response)
{
  debugPort.println("Connected");
 
  //mqtt.subscribe("/topic/0"); //or mqtt.subscribe("topic"); /*with qos = 0*/
  //mqtt.subscribe("/topic/1");
  //mqtt.subscribe("/topic/2");
 
  mqtt.publish("/test01", "ASDFG 313");

}
void mqttDisconnected(void* response)
{

}
void mqttData(void* response)
{
  RESPONSE res(response);

  debugPort.print("Received: topic=");
  String topic = res.popString();
  debugPort.println(topic);

  debugPort.print("data=");
  String data = res.popString();
  debugPort.println(data);

}
void mqttPublished(void* response)
{

}
void setup() {
  Serial.begin(19200);
  debugPort.begin(19200);
  esp.enable();
  delay(500);
  esp.reset();
  delay(500);
  while(!esp.ready());

  debugPort.println("ARDUINO: setup mqtt client");
  if(!mqtt.begin("UNIT01", NULL, NULL, 120, 1)) {
    debugPort.println("ARDUINO: fail to setup mqtt");
    while(1);
  }

  debugPort.println("ARDUINO: setup mqtt lwt");
  mqtt.lwt("/lwt", "offline", 0, 0); //or mqtt.lwt("/lwt", "offline");

/*setup mqtt events */
  mqtt.connectedCb.attach(&mqttConnected);
  mqtt.disconnectedCb.attach(&mqttDisconnected);
  mqtt.publishedCb.attach(&mqttPublished);
  mqtt.dataCb.attach(&mqttData);

  /*setup wifi*/
  debugPort.println("ARDUINO: setup wifi");
  esp.wifiCb.attach(&wifiCb);

  esp.wifiConnect("SSID","password");

  debugPort.println("ARDUINO: system started");

 
}

void loop() {

  esp.process();
  if (wifiConnected){
  
  }
}
 `

frugallabs, 2016-04-08:
You can call this in void loop
if (wifiConnected){
mqtt.publish("/test01", "ASDFG 313");
delay(5000);
}

tolbier, 2016-04-08:
Thank you. I will try it.....