tuanpmt/espduino #56: Getting Garbage after esp.ready()

nirajbhatt7, 2016-07-24:
Can someone please help.
I am getting garbage after esp.ready()

Output:
ARDUINO: Starting Device
ARDUINO: Waiting for ESP!!!
_â‚º[ØÐPŠþX¹a‚ç·âÌþ8�+ˆ¢·A¢ÃâªšF¢ªÂ«PT"p©ÅS0Ø8

-------------------------------------------------------------------------------------------------

Code Sample:
`
#include <SoftwareSerial.h>
#include <espduino.h>
#include <mqtt.h>

#define esp8266reset        A5                      //arduino pin connected to esp8266 reset pin (analog pin suggested due to higher impedance)

SoftwareSerial espPort(9, 10); // RX, TX
ESP esp(&espPort, &Serial, esp8266reset);

MQTT mqtt(&esp);
boolean wifiConnected = false;

.
.
.
.
.
.

void setup() {
  Serial.begin(19200);
  espPort.begin(19200);

  Serial.println("ARDUINO: Starting Device");
  
  esp.enable();
  delay(500);
  esp.reset();
  delay(500);

  delay(1000);
  
  Serial.println("ARDUINO: Waiting for ESP!!!");
  while(!esp.ready());
  
  Serial.println("ARDUINO: ESP is Ready");

`

nirajbhatt7, 2016-07-24:
Also note that my CHIP_PD is on A5 as I have a PCB that I designed last year and it was all working at some point in 2015 but now that I started playing with again, it has stopped working. Just FYI, I have taken latest for both and flashed ESP-01 with latest bin as well. My ESP-01 is 2 years old in case if that makes any difference

patmalcolm91, 2016-10-01:
I'm also having the exact same issue. Did you ever figure out a solution?